seho85/klipper-dgus #46: Gcode user macro 2 funktioniert nicht

Robnex, 2022-09-01:
Hi,
Laut evil666 funktioniert der Zugriff auf macro 2 nicht. Muss ich selbst noch validieren. 
Chat auf Discord in Display
Kanal

seho85, 2022-09-02:
I tried to reproduce the reported, but for me both *DGUS_USER_POSx* are called properly.

Maybe there is a problem with the user defined macro.

For the test I just used macros with no other content then printing out the comment, which are just printed at the console.

```
#Move Printhead to user defined position 1(Homing should be performed before in needed)
#Note: This a maintence position
[gcode_macro DGUS_USER_POS1]
gcode:
    {action_respond_info("DGUS Request: User Position 1 - Fill with proper commands")}
    

#Move Printhead to user defined position 2 (Homing should be performed before in needed)
[gcode_macro DGUS_USER_POS2]
gcode:
    {action_respond_info("DGUS Request: User Position 2 - Fill with proper commands")}
```

Pressing both Buttons leads to to:

[image]

So everything seams to be fine.

evilpeter666, 2022-09-03:
Hi seho85,
in der Konsole sieht das bei mir auch so aus. Aber wenn ich dort ein Kommando mit reinpacke, wird es nicht ausgeführt. Habe dazu einfach mal das Kommando aus Pos1 kopiert. Es passiert am Drucker aber nur bei dem Macro für Pos1 etwas, bei Pos2 kommt nur die Ausschrift in der Konsole.

seho85, 2022-09-04:
If you see the outputs, then macro is called.

The GCODE Commands needed to be properly indented that they are working. Maybe that the problem.

```
[gcode_macro DGUS_USER_POS2]
gcode:
    {action_respond_info("DGUS Request: User Position 2 - Fill with proper commands")}
    SET_PIN PIN=LED VALUE=1.00            ;LED Printhead On
    G4 P2000                              ;Wait 2000ms
    SET_PIN PIN=LED VALUE=0.00            ;LED Printhead Off
    G4 P2000                              ;Wait 2000ms
    SET_PIN PIN=LED VALUE=1.00            ;LED Printhead On
    G4 P2000                              ;Wait 2000ms
    SET_PIN PIN=LED VALUE=0.00            ;LED Printhead Off
```

If I use the macro above I get:

* the printing in the console
* LED on print head on for 2 seconds
* LED on print head off for 2 seconds
* LED on print head on for 2 seconds
* LED on print head off

evilpeter666, 2022-09-04:
Ich habe keine Idee, aber auch das funktioniert nicht. Die Konsole gibt die Info "DGUS Request: User Position 2 - Fill with proper commands" aus, aber es passiert sonst nichts.

Robnex, 2022-09-10:
Hallo EvilPeter,

Wir haben gerade nochmal an 2 Vyper den button getestet. Dieser ist voll funktionsfähig **wenn:**
1. Das Macro erfolgreich eingebunden ist
`[include ./dgus_display/dgus_display_macros.cfg]`
2. Das User_Macro zwei auch mit einem Befehl gefüllt ist.

Standardmäßig sieht es so aus:
(Einfach mal ein G28 hinzufügen)
`[gcode_macro DGUS_USER_POS2]
gcode:
    {action_respond_info("DGUS Request: User Position 2 - Fill with proper commands")}`

evilpeter666, 2022-09-10:
Hi Robnex,

ich kann mich leider nur wiederholen, es passiert nicht außer die Ausschrift in der Konsole. Was soll ich denn jetzt noch machen?
G28 habe ich z.B. eingefügt, es wird aber nicht ausgeführt....gib mir doch mal deine cfg Datei...dann probiere ich es damit...

seho85, 2022-09-10:
Welche Datei hast du denn in die *printer.cfg* importiert? Das aktuelle Setup installiert aufgrund eines Fehlers diese Datei zweimal.

Einmal in das *dgus_display* und einmal in *dgus_display/config*

evilpeter666, 2022-09-10:
Ich habe die cfg jetzt nochmal komplett neu erstellt, gleiches Ergebnis. Da muss doch an einer anderen Stelle der Fehler sein?!

evilpeter666, 2022-09-10:
> Welche Datei hast du denn in die _printer.cfg_ importiert? Das aktuelle Setup installiert aufgrund eines Fehlers diese Datei zweimal.
> 
> Einmal in das _dgus_display_ und einmal in _dgus_display/config_
> 
> Wichtig ist das du die Anpassung in der _dgus_display_macro.cfg_ vornimmst die auch per [include ...] in deiner _printer.cfg_ eingebunden ist.

Ok, das kontrolliere ich nochmal...so hatte ich das aber nie verstanden

evilpeter666, 2022-09-10:
Also das war ja ein großes Missverständnis. Das die cfg Datei da nochmal in dem anderen Ordner direkt zum anschauen liegt, war mir nicht bewusst. Ich kenne ja die Struktur von Klipper nicht wirklich. Nun ist es mir klar...Man Oh Man, das war ja ne schwere Geburt...Jetzt funktioniert es natürlich....

seho85, 2022-09-10:
Ok, super, das ist ja das wichtigste. 

Dann mach ich hier zu :)

closed 2022-09-10 by seho85