seho85/klipper-dgus #47: Cancel macro wäre ein anderes

Robnex, 2022-09-01:
Hallo,
Laut evil666 wäre das cancel Macro ein anderes zu dem auf dem Dashboard. Muss ebenfalls noch geprüft werden. Chat auf dem Discord in der Display Gruppe

seho85, 2022-09-02:
The display doesn't call any macro for canceling the print.

It calls the [Cancel Print Command](https://moonraker.readthedocs.io/en/latest/web_api/#cancel-a-print) defined in the Moonraker API.

If using this functions doesn't work for performing all actions that should have been performed by the MACRO which can be used to cancel the print, we need to have a look where the actual difference is.

seho85, 2022-10-22:
Due no response in the past few weeks, i'll consider this issue as solved.

Please post back if not.

closed 2022-10-22 by seho85