seho85/klipper-dgus #48: Verbesserung: Hinweis auf homing bei ztilt

Robnex, 2022-09-01:
Hallo,
Es wurde gefragt ob es möglich wäre einen Hinweis auf den homing Status beim ztilt auszugeben. Eventuell ein homing Button der grün ist oder so

seho85, 2022-09-02:
Informing the user about the homing state by turning the Homing Button green or sth. like that can be implemented in a later version.

For know I will change the base implementation of the *[DGUS_ZTILT]* to:

```
[gcode_macro DGUS_ZTILT]
gcode:
    {action_respond_info("DGUS Request: Perform Z-TILT")}
    {% if printer.toolhead.homed_axes != "xyz" %}
        {action_respond_info("Printer not homed.. performing G28 (homing before)")}
        G28
    {% endif %}
    Z_TILT_ADJUST
```

That change will do an homeing before the executing the *Z_TILT_ADJUST* before, if needed.

seho85, 2022-09-10:
Added homing if needed before z-tilt as default action in *dgus_display_macros.cfg*

See 0e47bf2ea7e8a0e5b98d782b5b6b649d505a4f8b

closed 2022-09-10 by seho85