seho85/klipper-dgus #49: Verbesserung screensavers

Robnex, 2022-09-01:
Wurde gewünscht. 
Sollte ja mit dem nächsten Update kommen

seho85, 2022-09-02:
A config that reduces the brightness of the backlite after an idle time of 60 seconds, was included in commit https://github.com/seho85/klipper-dgus/commit/ee2a0ac8782cc3ad595c83c638e4787d871fa576

Not sure what is meant with "improvement" in this case.

Currently the backlite should work like:

from ee2a0ac8782cc3ad595c83c638e4787d871fa576

"Implemented Backlight dimming and added documentation. Backlight is now 100% when not idle,
60 seconds after the last touch the backlight is set to 10% (sleep mode)
display needs be woken up by a touch operation on the display
(this operation doesn't execute any commands)"

seho85, 2022-09-18:
Closing this issue - backlight dimming integrated - no further responses here.

closed 2022-09-18 by seho85