seho85/klipper-dgus #53: Use Vyper mainboard serial port instead of Pi's

daduke, 2022-09-03:
hey,

first off: fantastic work, I had been trying to come up with something like this based on [this project](https://github.com/Desuuuu/DGUSPrinterMenu), but I never found sufficient time to suffer through the DGUS tool. Can't wait to try it on my Vyper (currently travelling). 
Based on Desuuuu's work it should be possible to use the Vyper mainboard serial port instead of the Pi one (I was able to talk to the screen and display stuff), so maybe this is something you'd like to look into - it would make using your project a whole lot easier..
[This screenshot](https://daduke.org/junk/vyper-klipper-menuconfig.png) shows the menuconfig settings for compiling Klipper for the Vyper with UART support for the screen.

cheers,
-Christian

seho85, 2022-09-03:
Thanks, it's always nice to hear that someone out there is appreciating my work.

I also saw the work that was made by desuuuu before I started the whole project. 

I choose the way of using the serial port on the Pi (Klipper Host Machine) because, at this point nothing (which exists in klipper mainline source) is available for transporting the serial data over klipper to a klipper mcu. I just want to stick to the mainline source of klipper, so that I doesn't need to maintain another fork of klipper. 

I've done this until right now to get my self created Display working (https://github.com/seho85/RaspberryPicoKlipperHMI) - which is now replaced by stock DWIN display.

But in between I talked to another klipper developer (on discord) which is currently developing: "A transparent UART bridge". This feature is exactly what is needed to get the display running directly connected to the Vyper Mainboard. It would allow to use the serial port on a klipper mcu over the klipper host machine.

When this feature is ready and has made it upstream into the klipper sources, then i will give this a try. 

Unfortunately the requirements to bring code upstream into klipper source is pretty high, and their accepting of pull requests is (from my personal view) so kind of conservative. So I personally decided just to use the upstream sources of klipper and follow the KlipperScreen approach of using the Moonraker API to talk to klipper.

daduke, 2022-09-03:
fair enough, thanks for the fast response. I'll try the current configuration when I get home.

seho85, 2022-09-03:
Lets just keep this issue open. And when "Transparent UART Bridge" is implemented in klipper we see how to continue.

Thanks for your contribution.

Gamerou, 2024-04-01:
Are there any updates?