seho85/klipper-dgus #61: DGUS Project folder almost works on Anycubic Kobra

kuuhgle, 2022-12-12:
I tried installing the whole project for my Anycubic Kobra, and the display connects and receives touch input, just the UI is zoomed in and tilted 90 degrees, the bitmaps are also not displaying.
Would be very nice to make it work for the Kobra, as most of it is working.

seho85, 2022-12-14:
The display orientation is controlled by T5LCFG.CFG

https://github.com/seho85/klipper-dgus/blob/master/dgus_project/DWIN_SET/T5LCFG.CFG

Maybe tweaking the config help.

A listing of the configuration options can be found in the Application Development Guide: https://github.com/seho85/python-dgus/blob/master/doc/source/vendor/T5L_DGUSII%20Application%20Development%20Guide.pdf

Starting at Page 20

I got no access to an AC Kobra. So I can't try by my own.

kuuhgle, 2022-12-15:
I managed to get the configuration right by editing that, but the scaling of the GUI is still wrong sadly. Do you have any idea where that is stored?
Weirdly, the scaling is zoomed in by about 300%, but the Displays are almost the same size, which doesn't make a lot of sense

seho85, 2022-12-15:
Maybe the display resolution is different (configuration file/hardware). The Vyper Display has 480x800. The graphics flashed to the display are raster graphics, so the *.icl files contains graphics scaled to 480x800.

You can try to find out which display is used by tearing it down. That's what i've done with mine to know which display im dealing with. On mine was a product label which used to look up the specs on the DWIN website.

Unfortunately AC doesn't provide anything for the Kobra Display to download.

Maybe you can find something in the Marlin sources for your Kobra by searching trough the code for the _ANYCUBIC_LCD_DGUS_ define. Which is set in the configuration.

gifter77, 2022-12-25:
FYI Anycubic does provide the display DWIN_SET folder for Kobra on their website, which links to a Google Drive:
https://drive.google.com/file/d/1s8MgWhRIj92Gd1T8nQiPWP1RVIAvqiAG/view

There is even a project that changed the LCD UI:
https://github.com/jojos38/anycubic-kobra-dark-ui

gifter77, 2022-12-25:
The Kobra screen is 272 × 480 FYI, so a much smaller resolution than the Vyper display. So I think you would need to recreate the entire DWIN project if you wanted this to work on Kobra screen.

cscomputers, 2023-02-05:
Hi, searching for a solution for my display, I've tried Desuuu's solution, but not worked, the display is not the same model, I have a Sovol SV01 Pro, comes with creality 4.2.2 board and DMG48270C043_03W display (T5L1), I flashed with Kobra's display FW and works ok, because is the same display.

I searched more and find this solution for CR6_T5L (same display size from T5L1), the guy have changed some files from DEGUUUUS and worked for him.
I've tried to flash this FW but is not worked. 

Can someone try to make similar job ? The screen size is the same, maybe to be more easy to do this.

Here the link of repo:
[](https://github.com/Thinkersbluff/DGUS-Reloaded_for_CR6-DWIN-SET_Component/tree/Refactored_For_CR6_T5L_DWIN_Display)

Thanks in advance!

BR.

Fragmon, 2023-07-29:
> Hi, searching for a solution for my display, I've tried Desuuu's solution, but not worked, the display is not the same model, I have a Sovol SV01 Pro, comes with creality 4.2.2 board and DMG48270C043_03W display (T5L1 DGUS2), I flashed with Kobra's display FW and works ok, because is the same display.
> 
> I searched more and find this solution for CR6_T5L (same display size from T5L1), the guy have changed some files from DEGUUUUS and worked for him. I've tried to flash this FW but is not worked.
> 
> Can someone try to make similar job ? The screen size is the same, maybe to be more easy to do this.
> 
> Here the link of repo: [DGUS-Reloaded_for_CR6-DWIN-SET_Component](https://github.com/Thinkersbluff/DGUS-Reloaded_for_CR6-DWIN-SET_Component/tree/Refactored_For_CR6_T5L_DWIN_Display)
> 
> Thanks in advance!
> 
> BR.

You can have a look on our repo. Its finished in the next days. You can find documentation here:
https://book.cryd.de/books/klipper/page/youtube-kobra-display-mit-klipper
https://github.com/cryd-s/kobra_extended/tree/main/klipper-dgus

Xilebo123, 2026-10-06:
The zoomed/tilted UI usually means the firmware is being loaded for a different display resolution or rotation than what your Kobra's actual DWIN panel expects. The Anycubic Kobra ships with a specific DWIN variant (check the label on the back of the display for the full model number, e.g. DMGxxxxxCxxx_xxx).

What helped me when I hit the same issue: pulling the exact datasheet for my specific model to confirm the native resolution, rotation, and touch calibration parameters before flashing. I put together a searchable catalog of all 382 DWIN display models with direct datasheet PDF links: https://dwin-hmi.eu — just type in your model number and you get the spec sheet in one click. Independent community project, not DWIN-affiliated.

Once you have the right datasheet, the rotation and resolution settings in the DGUS project config should line up. Hope that gets the Kobra display rendering correctly!