seho85/klipper-dgus #69: Show thumbnail from code when printing?

matphillips, 2023-10-11:
Fantastic project, just what I was looking for. I recently upgraded to Klipper and was disappointed when I realized the screen on the Vyper was now useless. This gives it a purpose again.

Now, I see an open issue regarding using the existing connection to the screen via the printer control board, and that would be a much needed improvement, essential really, as it would turn the screen off when the printer is off. I fully agree with not wanting to use a custom fork of Klipper for this, so I will wait with optimism!

So, on to the issue, would it be possible to pull the png thumbnail that is now commonly included in the gcode file and display it on the screen?

seho85, 2023-10-11:
Thanks for the kudos.

Turning on and of the screen with the printer could not be easily achieved - there is simply no option to power down the screen by software. Reducing the brightness of the background illumination (to 0%) should be achievable.

If you want the display to be completely powered off the easiest way, would be to power the display over the printers psu. So use the 5V and Ground to which the Display is usually connected to power the display. And connect the RX and TX Pins to the Pi (or whatever Klipper Hostmachine you are running, I just call in Pi in my answer :D). GND should also be connected between PI and Printer Mainboard, could be, that is also working without the connection but a common GND is usually needed.

Also using a relay (5V / Normal Open) for switching the power of the Display could be an option. The relay just acts as a switch on the 5V power line to the display. Relay is switched by the 5V supply on the printer. When printer is switched on (provides 5V) the relay is closed, when printer powered off relay opens, display powers off.

Maybe something can be done using moonraker, a GPIO, a realy or a transistor to switch the powersupply for the display. I think klipper can't be used for this, when the Pi loses the connection to a MCU associated with your printer, it can't change GPIOs anymore.

About the thumbnail:

For showing the thumbnail on the screen, the thumbnail has to be transmitted to the screen. But the DGUS Display used here is not utilized like a "normal" monitor (it doesn't get the information draw pixel at X=0 and Y=0 to RGB Value). The graphics shown on the Display are stored on the Display (by flashing the Display) and only the variable Data is exchanged. (Text, Temperatures, Numbers, which mask should be shown, a.s.o).

I saw that the manual states something about "drawing" on the Display using control interface, but I never stepped deeper into that. Unfortunately the manual is not very clear at this point if it is possible to transmit a PNG (or some other kind of raster graphic). I gave it a try but haven't figured out how this is working.

If someone provides a working example how to achieve this showing a raster graphic on the display, i could integrate that to my klipper_dgus project.

Currently i'm not focusing too much on 3D printing, the 3D printer become a handy tool for my other projects, but the printer is self is currently no project i'm actively working on. Maybe in the upcoming winter month I found more time to focus back on my 3D printer.

matphillips, 2023-10-11:
Thanks for the reply.
I didn't realize the DGUS screen was quite that limited, so yes it makes sense that showing the thumbnail is either impossible or way harder than it should be.

closed 2023-10-11 by matphillips