seho85/klipper-dgus #71: Is it possible to hook this screen up to Klipper screen instead?

SejeAnd, 2023-12-27:


seho85, 2023-12-27:
No, you can't put raw display on the DGUS display.

See https://github.com/seho85/klipper-dgus/issues/69#issuecomment-1757856067

thghca, 2023-12-27:
I have a crazy idea. It is possible to make a pixel font. This way you can display at least a black and white image.

seho85, 2023-12-27:
> I have a crazy idea. It is possible to make a pixel font. This way you can display at least a black and white image.

Feel free to do so :) 

I would be happy to see a working prove of concept. According to the manual "drawing on the display" should be possible, i think thats could be the easier way to achieve the same goal. But i never got this working.

 Currently working on this project went out of my scope. My time is limited and currently other projects have priority.