seho85/klipper-dgus #77: Klipper DGUS does not yet work

gsohler, 2025-02-27:
Hi Sebastian,

1st of all, thank you for your efforts.

I followed your instructions, assembled and connected the cable updoaded the firmware, installed the software, 
but the Klipper-display doe snot yet work

*  Is it ok, that on printer startup, there is still this Anycubic Animation ?
*  Flashing of the display firmware appeard to be ok. there were 20 lines with 0000 reports, but it it finished in less than 0.5 sec

* changed the corect serial port and ./dev/usb/by-id has shown /dev/ttyUSB0 which appears to be to the "good" one

* can i use minicom to perform some communicatoin tests ?
* In Journal i see many messages like :   
* 
* eb 27 00:30:06 klipper systemd[1]: klipper_dgus.service: Failed with result 'exit-code'.
Feb 27 00:30:06 klipper systemd[1]: klipper_dgus.service: Scheduled restart job, restart counter is at 3.
Feb 27 00:30:06 klipper systemd[1]: Stopped DGUS Display for Klipper.
Feb 27 00:30:06 klipper systemd[1]: Started DGUS Display for Klipper.
Feb 27 00:30:06 klipper systemd[3386]: klipper_dgus.service: Failed to determine user credentials: No such process
Feb 27 00:30:06 klipper systemd[3386]: klipper_dgus.service: Failed at step USER spawning /usr/bin/sleep: No such process
Feb 27 00:30:06 klipper systemd[1]: klipper_dgus.service: Control process exited, code=exited, status=217/USER

* had to change "pi"  with gsohler in , /etc/systemd/system/multi-user.target.wants/klipper_dgus.service 
 but still no success

* i manually start: home/gsohler/klipper-dgus/venv/bin/python3 /home/gsohler/klipper-dgus/src/main.py -c /home/gsohler/printer_data/config/dgus_display,  there is lots of log, but display does not change

* at least the cable appears to be correct, display receives power

what could be the issue ?

CybrGost69, 2025-04-03:
seems this is only for the DWIN T5L chip

gsohler, 2025-04-03:
sorry no,
got it working in the meantime
isues were:

* did not manage before to flash the new image
* broken cable.

after both issues were resolved, now its working
thx

closed 2025-04-03 by gsohler